flagxor/ueforth #1: Compile fail

arblake, 2021-01-12:
Brad,
I get an immediate compile failure:
arduino.template:1:1: error: expected unqualified-id before '{' token
 {{opcodes}}
 ^

arblake, 2021-01-17:
Now I have this error when trying  make:
make: *** No rule to make target 'arduino/arduino.template.ino', needed by 'out/arduino/ueforth/ueforth.ino'.  Stop.

arblake, 2021-01-19:
Compiled ok
When I try to start the wifi I get this :
--> wifi.begin
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

flagxor, 2021-02-08:
Wifi.begin requires some parameters, see:
https://eforth.appspot.com/

z" SSID" z" PASSWORD" WiFi.begin

Unfortunately, I have not yet found a good way to add runtime memory guards. ESP-32 doesn't have a real mmu, so anything will have a runtime cost.

closed 2021-02-08 by flagxor