flagxor/ueforth #6: more documentation please

arblake, 2021-03-03:
Brad,
I need more help and examples for setting up the web server and pages.

Adrian

flagxor, 2021-06-29:
Sorry for the slow reply.
Web server is a bit in flux.
There's two approaches lightly documented at https://eforth.appspot.com/
The WebServer interface mirrors arduino's WebServer.h
Alternatively, in v7.06+ I've added httpd, which implements a web server directly in sockets.
Both have some issues and limitations, but my hope is to deprecate the older WebServer interface at some point.
For current examples check out these:
https://github.com/flagxor/eforth/blob/main/ueforth/esp32/camera_server.fs
https://github.com/flagxor/eforth/blob/main/ueforth/posix/web_interface.fs

Apologies for the mess :-)

-BradN

bob-g4bby, 2021-07-12:
Also see quite a good glossary of words (I'm the author!) at [https://docs.google.com/spreadsheets/d/1UMBc05BkfW7QapvyUZYLIYJR7FAVHONnCJO_BEdAqCg/edit#gid=0](url)

MPETREMANN11, 2022-01-23:
Good evening,
On my personal site, I am working to set up documentation as complete as possible regarding ESP32Forth:
[](https://esp32.arduino-forth.com/index/glossaire)

MPETREMANN11, 2022-08-04:
> Also see quite a good glossary of words (I'm the author!) at [https://docs.google.com/spreadsheets/d/1UMBc05BkfW7QapvyUZYLIYJR7FAVHONnCJO_BEdAqCg/edit#gid=0](url)
Broken link: "No results matched your search."

bob-g4bby, 2022-08-05:
Also see quite a good glossary of words (I'm the author!) at - should have been:-
https://docs.google.com/spreadsheets/d/1UMBc05BkfW7QapvyUZYLIYJR7FAVHONnCJO_BEdAqCg/edit#gid=0
The words are grouped by topic e.g. STACK, MATHS etc. so that the new programmer only has a little list to find the word he wants. I am currently trying to figure out how to use the SOCKETS vocabulary - but lack of examples makes it a tall order. Were examples written for test purposes or is this vocabulary still untested? I've written and shared quite a bit of code for ESP32FORTH, but quite a few vocabularies remain too cryptic to be usable, which is a shame. At the same time I quite understand this is the work of one generous man and the project is still in a state of flux.

MPETREMANN11, 2022-08-05:
> cryptic to be usable, which is a shame. At the same time I quite understand this is the work of one generous man and the project is

Good Job.

MPETREMANN11, 2022-08-05:
> > cryptic to be usable, which is a shame. At the same time I quite understand this is the work of one generous man and the project is
> 
> Good Job. On my web site, the documentation is often completed by examples. The documentation is in french and english https://esp32.arduino-forth.com/index/glossaire [image]

And I use highlithing the FORTH source code (writed in PHP). It's explained here:
https://esp32.arduino-forth.com/article/highlightFORTHwithPHP

All highlighted code give access to online documentation....

MPETREMANN11, 2022-08-05:
> 

I'm reading the Serial Communication. It's not complete. You omit Seria2.xxxxxx words. If you want to complete your documentation, I suggest you take the explanations on my site without worry.

bob-g4bby, 2022-08-07:
Hi Marc,
I would like to write some example programs using the SOCKETS vocabulary. I see a small example on your web page Understanding ESP32Forth Socket Management (arduino-forth.com). Do you have more? I would use the test and measurement language Labview on the PC to send and receive arrays of data, then write it up in a small paper. It seems a shame that users cannot transfer high speed data between esp32forth and a PC e.g. audio. The lack of documentation for certain vocabularies holds us back. I shall try to contact Bradley Nelson about it.
Regards, Bob

MPETREMANN11, 2022-08-07:
> Hi Marc, I would like to write some example programs using the SOCKETS vocabulary.......
My mail for direct contact:   petremann(at)arduino-forth.com