flagxor/ueforth #7: Issue with compiling (eForth) under Arduino

techniks, 2021-03-13:
Nice job by the way!
(Checked out your french bread formula .. going to try your ratios)
2 basic things I don't understand (but I'm good at making bad assumptions)
The {{opcodes}} ,{{calling}}, {{core}}, etc. I assume are macros to to be filled in automatically by some external data? At this point I REM'd them out. (They cause compiler error.)
Also "cell_t" errors with "not defined in this scope."
I copied "typedef uintptr_t ucell_t;" and " from esp32forth file. Seems to have fixed this.
Then finally,  "'g_sys' was not declared in this scope."
Can't seem to fix this.

I can't figure out what I'm not seeing.
It would be wonderful if you could help me out.
Thanx,
Frank

flagxor, 2021-06-29:
Hi Frank,

Sorry for the very slow reply.
The core is build using X-Macros:
https://en.wikipedia.org/wiki/X_Macro

I'm unsure from the errors you're getting, but perhaps you caught a bad rev, or aren't using the gnumake build?
When building directly from this repository you'll need to use the Makefile under ueforth/
Alternatively, if you're targeting Esp32 only, I've got post script munged snapshots of the source code at: https://esp32forth.appspot.com/

What platform are you trying to build for?

Regards

-BradN

frankcappelle, 2025-02-18:
Brad, I'm sure your very busy, so I appreciate any insight you can give me.
I'm interested in using ueForth eb as a graphical (canvas) scripting language within a web page.
I'm developing a project that utilizes esp32forth on a esp32 in which I'm sending commands from a webpage using web serial. 
I want to send the same command to the the local web canvas so that the same image will appear on the hardware device and the web page. I would write the appropriate forth words for both devices/platforms.
My issue comes down to how I can inject forth words into the ueForth directly (through javascript calls???) and eliminate the on screen terminal. Any insight would be very helpful.
Thanks, Frank