flagxor/ueforth #9: How do you use local variables or local variables are very broken

craiglindley, 2021-07-10:
First let me say ESP32forth is a great tool which I am really looking forward to using in many projects.
Also, I am a big fan of named parameters and named local variables which make Forth a whole lot easier to program in.

Maybe I don't understand how to use locals in ESP32forth so my problems might be of my own making. Here is a short test that I cannot get to compile:

: bytearray ( size -- ) ( i -- addr ) create allot does> + ;

\ Define an array of pixel values
\ Three bytes are required for each pixel
10 3 * bytearray pixels

\ Convert indiviual RGB color components into 24 bit color 
: color24  { r g b }
  r 16 <<
  g  8 <<
  b 
  or or
;

\ Set pixel using a 24 bit color value
: setPixelColor  { color24 pixelNum }
  color24 16 >> 255 and
  color24  8 >> 255 and
  color24       255 and
  pixelNum setPixelRGB
;

My issues with locals, besides the above not compiling, are:
1. if you do something like { color24 pixelNum --- } , they don't work
2. If you define some locals and use them in a do .. loop, they don't work.
3. You cannot store to them or change their values

If I am using locals wrong, please enlighten me. In the past I have used pForth which has both named parameters and local variable support which was rock solid. That might be a good model to use in ESP32forth.

flagxor, 2021-07-12:
Hi!
Thanks for catching a very tricky bug!
It turns out in addition to the partial locals support, you ran into a thorny alignment issue that was tickled when using both 8 character names and locals.

I've also used this as an opportunity to fix up the locals support which had some gaps (previously didn't support multiple uses or use within a DO LOOP). I've also added support for TO and +TO.

I've just published v7.0.6.4 and v7.0.5.4 (stable) that include the fixes.

You should be good with the above now.

Cheers

-BradN

closed 2021-07-12 by flagxor

craiglindley, 2021-07-12:
Wow thanks very much for this, I have a lot of forth code that uses locals
that I won't have to rewrite

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

craiglindley, 2021-08-05:
Did you see the new issue with locals that I wrote up? It could have
something to do with 8 character names as well.

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.