flagxor/ueforth #17: Extending UART to UART2, possible?

MPETREMANN11, 2022-02-12:
Hello,
Thank you first for your great work on ESP32Forth
Do you plan to extend UART to UART2 on ESP32Forth?
In my developments, I use this UART2 port thanks to this modified code: 

#define REQUIRED_SERIAL_SUPPORT \
  X("Serial.begin", SERIAL_BEGIN, Serial.begin(tos); DROP) \
  X("Serial.end", SERIAL_END, Serial.end()) \
  X("Serial.available", SERIAL_AVAILABLE, PUSH Serial.available()) \
  X("Serial.readBytes", SERIAL_READ_BYTES, n0 = Serial.readBytes(b1, n0); NIP) \
  X("Serial.write", SERIAL_WRITE, n0 = Serial.write(b1, n0); NIP) \
  X("Serial.flush", SERIAL_FLUSH, Serial.flush()) \
  X("Serial2.begin", SERIAL2_BEGIN, Serial2.begin(tos); DROP) \
  X("Serial2.end", SERIAL2_END, Serial2.end()) \
  X("Serial2.available", SERIAL2_AVAILABLE, PUSH Serial2.available()) \
  X("Serial2.readBytes", SERIAL2_READ_BYTES, n0 = Serial2.readBytes(b1, n0); NIP) \
  X("Serial2.write", SERIAL2_WRITE, n0 = Serial2.write(b1, n0); NIP) \
  X("Serial2.flush", SERIAL2_FLUSH, Serial2.flush())

This code has been tested and verified with communication to a LoRa transmitter.
cordially

flagxor, 2022-02-28:
Good idea. I've added it, and will push out in next release.

closed 2022-02-28 by flagxor

PythonLinks, 2023-12-31:
This looks most useful.  I need two uarts for debugging a mecrisp ice cpu on the pico-ice. 
I could not find the command to talk to the uart.  What is it?
Does the Pico-Ice also have two uarts which i can use to talk to the FPGA?  If so what are the pin numbers?

Maybe I just am not able to find the correct documentation.  Is there a page I should be reading?

For context, the default pico-ice firmware has just 1 uart to talk to the fpga.  Reading the source code, It was not obvious what I need to do to add a second one. 

And microPython is even larger, scarier, and does not include the pico-ice sdk. 

"uEForth, you are my only hope"  -Princess Leia