flagxor/ueforth #46: timers error....

MPETREMANN11, 2023-03-26:
Hello,

timers
: test1 ." hello" cr ;
' test1 1000000 0 interval

it fails with this error:

E (85734) timer_group: timer_isr_register(262): HW TIMER NEVER INIT ERROR

MPETREMANN11, 2023-03-27:
Good morning,
I searched on my side.
I have several ESP32 boards, some compiled with the old ESP32 library. It works....
To compile version 7.0.7.8+ you must install the latest ESP32 library. This library is the problem.
It is this word that causes the problem in the source code:

in code, line 790:

`static cell_t TimerIsrRegister(cell_t group, cell_t timer, cell_t xt, cell_t arg, cell_t flags, void *ret);`

and in line:

`  YV(timers, timer_isr_register, n0 = TimerIsrRegister(n5, n4, n3, n2, n1, a0); NIPn(5))`

This code hasn't changed in any ESP32forth versions I've picked up.

I don't have any more items at the moment.
It would be necessary to analyze the source code of the ESP32 library. It's very complicated.

vposelt, 2023-03-28:
Together with this timers problem I have in 7.0.7.x compiled with Arduino 1.8.19 and new libraries 2.0.1 up problem with hw interrupts PINCHANGE, which generates funny error " Guru meditation error..." I explained it more in detail on Facebook https://www.facebook.com/groups/esp32forth/ .

MPETREMANN11, 2023-03-28:
Good morning,
I tried to find the origin of this problem.
To compile version 7.0.7.9, you must install a new version of the ESP32 library.
Versions prior to 7.0.7.9 available in binary were compiled with the old ESP32 library.
The "HW TIMER NEVER INIT ERROR" issue is reported on other web pages unrelated to ESP32forth, including:
https://esp32.com/viewtopic.php?f=13&t=25202
The cause would be the absence of a call to timer_init()

MPETREMANN11, 2023-03-28:
https://esp32.arduino-forth.com/article
Problème en cours d'analyse.....
Problem being analyzed.....

MPETREMANN11, 2023-05-25:
Bug corrected

closed 2023-05-25 by MPETREMANN11