flagxor/ueforth #49: 100% CPU load

farvardin, 2023-04-12:
I've seen even when idle, the ueforth REPL is running at 100% CPU. Maybe it could be possible to detect if there is no activity and then be lighter on the CPU?

FoundOne, 2023-06-06:
I can confirm that the latest git version build just runs my laptop's CPU at 100% and does nothing. My specs are:
Void Linux x86_64-musl, NodeJS v18.16.0, GCC 12.2.0
Remark I have to link /usr/bin/node to /usr/bin/nodejs in order the build to work and the test suit nearly freezes my computer, because of the 100% CPU overload.

flagxor, 2024-02-24:
Oh, very good point.
In making KEY? I've ended up making it spin loop when waiting for a key.
Working on a fix...

flagxor, 2024-02-24:
Fix in eaa8a3c4c3ec00995ffc42873992b3dd89101d4f

closed 2024-02-24 by flagxor