flagxor/ueforth #53: Load cannot be called twice in one block

vkichline, 2023-06-05:
If block 1 of blocks.fb contains the following code, it works as expected:
`2 3 thru`
If, on the other hand, it contains the following, only block 2 is loaded:
`2 load
3 load`
Expected: both block 2 and block 3 should be loaded.

Why it matters: I want block 1 to be a system load block, essentially configuring my system. Typically I load several individual blocks and several series of blocks using load and thru repeatedly in block 1. This makes a Forth system very flexible.

flagxor, 2024-02-24:
Hi vkichline,

I'm having trouble reproducing this.
Which platform are you seeing this on?

vkichline, 2024-02-24:
Esp32Forth on a variety of chips. Load is just:  block 1024 evaluate, so if the loaded block contains a load, there is no way to continue at the next statement once the inner load is completed. gForth and others maintain a list to unwind.

vkichline, 2024-02-25:
Here is a proposed repo case:

Screen 1:
.” Screen 1” cr
2 load
.” Required: print this line.” Cr

Screen 2:
.” Screen 2” cr

———

Expected:
Screen 1
Screen 2
Required: print this line.

Actual:
Screen 1
Screen 2