flagxor/ueforth #61: ninja: which version? and/or missing D8

farvardin, 2024-01-26:
I can no longer build ueforth since the switch to ninja.

i get a 

```
./configure.py 
V8 checkout in $HOME/src/v8 not found, ignoring.
Windows %LOCALAPPDATA% not available, Windows support disabled.
Traceback (most recent call last):
  File "/temp/github/forth/ueforth/./configure.py", line 569, in <module>
    Main()
  File "/temp/github/forth/ueforth/./configure.py", line 555, in Main
    InitOutput()
  File "/temp/github/forth/ueforth/./configure.py", line 234, in InitOutput
    D8 = {D8}
NameError: name 'D8' is not defined
```

error

I've tried with python3.6 and python3.8. Which version is required. What is missing? what is this "v8" ?

farvardin, 2024-01-26:
maybe it's this?

https://www.npmjs.com/package/d8

znmeb, 2024-02-20:
Nope ... it appears to be a dependency that's installed on `flagxor`'s system but isn't listed in the build instructions in the README. `D8` is an internal variable to the build process. I think it's related to `npm` or `v8`, but I'm neither a Python nor a NodeJS developer, so I can't contribute a patch.

farvardin, 2024-02-21:
Let's hope @flagxor could help us about this...

znmeb, 2024-02-24:
Brad has fixed this ... there are still some missing dependencies but I'll open separate issues for those.

flagxor, 2024-02-24:
Sorry about that.
A bunch of the dependencies logic became more explicit with ninja.
I think I've got this fixed now. Please re-open if not!

closed 2024-02-24 by flagxor

znmeb, 2024-02-24:
Pull request is live.