flagxor/ueforth #65: multiple CR on the web

farvardin, 2024-04-08:
It looks like the CR command doesn't output on the web as it should, like on the terminal:

```
--> ." test" cr cr ." test2" cr 
test
test2
 ok
--> 
```

instead of outputting two carriage returns:

```
test

test2
```

I've tried to fix this in the code of a webpage, like that:

```
JSWORD: cr2 { } 
    var br = document.createElement('br');
    context.terminal.append(br);

~ 

forth definitions

only web also forth

: cr cr2 ;

```

it has some potential to work, but the problem is the line breaks (with 'br') are not placed where they are encountered, but they are left at the end of the last element. I've tried several things, but it didn't work better.

```
testtest2

(2 line breaks)
```

farvardin, 2024-04-08:
as a workaround, I've managed to get multiple ``cr`` if I redefine ``cr`` like that:

```
: cr space 13 emit nl emit ;
```

it works but it won't behave like the standard does (by adding an extra space at the end of a line). Maybe it might be sensitive to either fix it differently for the web, or to add this code only on the web version (not the desktop one)

flagxor, 2024-11-08:
Hi, I'm not sure if this got fixed incidentally from something else, but not able to reproduce.
Closing, please reopen if you're still observing it!

closed 2024-11-08 by flagxor