flagxor/ueforth #69: Error building with gcc 13.2.1

erlanger, 2024-05-08:
Using archlinux to build.

```
$ ninja posix
FAILED: out/posix/ueforth 
g++ -O2 -I . -I out -s -DUEFORTH_MINIMAL -fno-exceptions -ffreestanding -fno-stack-protector -fomit-frame-pointer -fno-ident -ffunction-sections -fdata-sections -fmerge-all-constants -Wl,--build-id=none -std=c++11 -Wall -Werror -no-pie -Wl,--gc-sections -s -Wl,--gc-sections -no-pie -Wl,--build-id=none posix/main.c -o out/posix/ueforth -ldl -MD -MF out/posix/ueforth.d && strip -S --strip-unneeded --remove-section=.note.gnu.gold-version --remove-section=.comment --remove-section=.note --remove-section=.note.gnu.build-id --remove-section=.note.ABI-tag out/posix/ueforth
In file included from /usr/include/c++/13.2.1/cmath:41,
                 from /usr/include/c++/13.2.1/math.h:36,
                 from ./common/floats.h:15,
                 from posix/main.c:22:
/usr/include/c++/13.2.1/bits/requires_hosted.h:34:4: error: #error "This header is not available in freestanding mode."
   34 | #  error "This header is not available in freestanding mode."
      |    ^~~~~
In file included from /usr/include/features.h:503,
                 from /usr/include/dlfcn.h:22,
                 from posix/main.c:15:
/usr/include/stdlib.h:740:38: error: ‘int at_quick_exit(void (*)())’ conflicts with a previous declaration
  740 |      __THROW __asm ("at_quick_exit") __nonnull ((1));
      |                                      ^~~~~~~~~
In file included from /usr/include/c++/13.2.1/stdlib.h:36,
                 from ./common/tier0_opcodes.h:17,
                 from posix/main.c:19:
/usr/include/c++/13.2.1/cstdlib:63:18: note: previous declaration ‘int std::at_quick_exit(void (*)())’
   63 |   extern "C" int at_quick_exit(void (*)(void)) _GLIBCXX_NOTHROW;
      |                  ^~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

farvardin, 2025-01-06:
I can confirm this problem on debian as well. I've modified build.ninja to force to use g++-11:

```
CXX = g++-11
```

tylerdq, 2025-03-10:
> I can confirm this problem on debian as well. I've modified build.ninja to force to use g++-11:
> 
> ```
> CXX = g++-11
> ```

Did this change get the build working for you? Now it fails again for me with `g++-11 not found`.

farvardin, 2025-03-10:
> Did this change get the build working for you? Now it fails again for me with `g++-11 not found`.

g++ is the tool for building ueforth. It's often included with the gcc packages. What is your OS for trying to build this? If it says "g++-11 not found", then install this version. Or use another one, different from gcc 13 which has issues.

It worked for me because I have gcc 11 installed on my system.

nonchip, 2025-07-08:
the actual solution (at least to build the `posix` version for linux) is to remove `-ffreestanding`.

tylerdq, 2025-07-13:
Removing `-ffreestanding` from `configure.py` resolved the issue for me on posix. Thanks!